tomaskovacik/vwcdavr #2: Atmega328

olii, 2016-09-26:
Does not work with Arduino nano(Atmega328) and Skoda Fabia Symphony.
I see data comming from head unit but on display it is "No CD Changer".

tomaskovacik, 2016-09-26:
I do not have skoda fabia with symphony.... but, which code? how you connect it to ? do you have pics ? did you connected it like  this https://github.com/tomaskovacik/hw/blob/master/kicad/arduino_based_cdc_emulator/connection_to_HU.png ?

olii, 2016-09-26:
no matter what project I used(https://github.com/tomaskovacik/vwcdavr/tree/master/CDC_emulator) , I read instructions in source how to connect to pins. As I wrote before, nano receives command from head unit, so I think connection is ok. However, Tx line on arduino(pin 13) is connected to LED, but I do not know if it can be the problem.

tomaskovacik, 2016-09-26:
Try vwcdpic if it's works if you can

---- Oliver Nemček wrote ----

tomaskovacik, 2016-09-28:
hello, so I am on computer now, can you send me output of serial console of arduino? it should tel there if it understend what radio send to "changer" if there is problem it will not work. also pictures how do you wire it up please (idealy to my personal email adress). thats all I can do for you. Or I can send you one of wvcdpic units for free if it's works and you can try to fix vwcdavr code to works with skoda symphony?

olii, 2016-09-28:
I used this source I found somewhere at russian site. It works fine and that is all I want. I dont want advanced features, key  press, minutes and seconds ... 

Maybe if you want, you can also use Spi library, because when I used my second arduino as logic analyzer, the output looked really bad. If I use this source, the output looks just fine.

Now I think I am going to build preamp from your scheme :)

https://gist.github.com/olii/56c874f41b649496672b34da8177ee7e

olii, 2016-09-28:
btw I think that emulator could decode radio commands, but radio couldnt because of some bug in the code

tomaskovacik, 2016-09-28:
if output look bad, that its problem with code/HW/freq, I will check it. Spi is ok when radio can handle the speed. I try it, but it's HW specific = not good for generic code which has to be ported to other mcus. Also code which you linked  work, but in some cases radio will not tolerate if you press button and cdchanger will not react as it should.

my be you did not configure it proparly. for basic functions you need to comment out mpd, bluetooth android stuff ... 

just to clarify:
I'm not author of original C code, you can try original one, port it to arduino by your self and then tell me what I f*ck in it :) it will help other guys,  link is in coment of each code (www.mikrocontroller.net/attachment/31279/vwcdc.c).

Shyd version is just my port for shyd code for arduino, shyd version is what is on russian forum, just striped from other "mpd" stuff, or its my port of his work striped down.

tomaskovacik, 2018-06-29:
2ys inactivity, closing

closed 2018-06-29 by tomaskovacik

kalabase, 2021-05-10:
after struggling for days mine finally went into CD Changer mode with Arduino IDE, UNO and a VW MK1 ZAR. thanks @tomaskovacik  if your have paypal for sponsor share link I can donate few bucks to show gratitude.
Next step is to connect audio signal
[image]
[image]
[image]