tomaskovacik/vwcdavr #3: Debugging 'dataerr' messages

justinvdk, 2017-12-14:
I've hooked up an Uno to my RCD 200. Everything seems to work okay (bit glitchy, but restarting radio + arduino usually does the trick).
To debug this I hooked up the serial monitor and it got spammed with `dataerr` messages.
Like so:
```
AudAudi Concert I Multimedia Gateway Ver.1.0c
RING
LIST2
80dataerr
Audi Concert I Multimedia Gateway Ver.1.0c
NEXT
LIST1
LIST2
dataerr
80Audi Concert I Multimedia Gateway Ver.1.0c
Audi Concert I Multimedia Gateway Ver.1.0c
Audi Concert I Multimedia Gateway Ver.1.0c
Audi Concert I Multimedia Gateway Ver.1.0c
LIST1
Audi Concert I Multimedia Gateway Ver.1.0c
LIST3
5300000000dataerr
LIST4
dataerr
80LIST5
400000dataerr
0000dataerr

```

Is this an actual error, or somewhat expected?

From the code the `dataerr` indicated a framing type data error, and also that "This should never happen on normal head unit sending 32 bit command strings with error free data".

I can't quite debug where the (seemingly) random numbers come from (`80`, `5300000000`, `400000`, etc). I.e. I can't find where in the code they are put into the txbuffer, via `EnqueueString` or otherwise.

Any ideas?

Ps. The CD / track buttons DO change the cd and track accordingly (cd1 - cd6 and TR01 - TR99).
Pps. Only thing hooked up are CLC, DATA_IN, DATA_OUT and ground. So without audio (Audio did work in a previous test, as long as the radio is in playing state, and I don't press buttons too quickly. If I do the radio stops playing).

tomaskovacik, 2017-12-14:
hi, try regural version (without just_8bit_timers)..

regural version use 16bit timer, and also ICP1 which stopes timer and run ISR, 8bit version was originaly made to port it all to ATtiny85 mcus, it works but as you can see it need to by clean up or debug. there is lot of done in ISR ... which is not good as we know :)

more here:  http://kovo-blog.blogspot.sk/2015/06/vwcdpic-for-avr-arduino-and-attiny.html

justinvdk, 2017-12-14:
Ah I see. I figured that they would both work good and chose 8-bit only to have the 16-bit for myself to use (planning to interface with some other device).

Will try soon!

tomaskovacik, 2017-12-14:
bluetooth od dashboard ?:)

justinvdk, 2017-12-14:
I got a [SparkFun Purpletooth Jamboree](https://www.sparkfun.com/products/11924). I should be fine as it is possible to communicate over Serial.

I saw the TODO for AT commands, unfortunately I won't (can't) be helping with that :(

I WILL try to make it somewhat nice but I'm not very experienced in programming embedded stuff, so most is new to me.

justinvdk, 2017-12-14:
Switching to the "normal" version (with 16-bit timer) does not spew out errors (yay!).

But (for me, at least) the `capbuffer` isn't filled with anything (stays `000000000000000000000000`).

I can't investigate right now but will later. I'll close this issue as the original problem is not relevant anymore.

closed 2017-12-14 by justinvdk