tomaskovacik/vwcdavr #5: Updated to Arduino 1.8.5 Still not compiling

JL01152, 2018-05-28:
Hi and thank you for your quick response. On my previous attempt I was using an outdated version of the Arduino IDE so I updated to 1.8.5 and now I'm getting the following error messages. This is for the  cdc_arduino_ported_k9spud_vwcdpic.ino file.

Arduino: 1.8.5 (Windows 7), Board: "Arduino/Genuino Uno"

C:\Users\WINDOW~1\AppData\Local\Temp\ccbR9Kq2.ltrans0.ltrans.o: In function `Init_VWCDC':

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:802: undefined reference to `ResetTime()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:804: undefined reference to `SetStateIdleThenPlay()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:813: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:815: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:817: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:819: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:856: undefined reference to `SendPacket()'

C:\Users\WINDOW~1\AppData\Local\Temp\ccbR9Kq2.ltrans0.ltrans.o: In function `main':

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:2156: undefined reference to `SetStateIdle()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:2183: undefined reference to `SetStatePlay()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:2190: undefined reference to `SetStateIdle()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:2206: undefined reference to `ResetTime()'

C:\Users\WINDOW~1\AppData\Local\Temp\ccbR9Kq2.ltrans0.ltrans.o: In function `CDC_Protocol':

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1438: undefined reference to `SendPacket()'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1469: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1471: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1473: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1529: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1539: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Windows7HD\Documents\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:1545: undefined reference to `ScanCommandBytes()'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tomaskovacik, 2018-05-28:
please copy whole directory, not just one .ino file. functions.ino file is file which contain missing functions.

t.

JL01152, 2018-05-28:
Thanks a million. I was never gonna get that sorted. I successfully uploaded and will confirm functionality. 
Closing.

closed 2018-05-28 by JL01152