tomaskovacik/vwcdavr #6: Cdc mode activated. No play function

JL01152, 2018-05-28:
I successfully compiled, uploaded and connected the arduino to audi symphony 2. It recognizes cdc1 track1 00:00 but does not emit sound from aux nor does it have play option. Is there sething I am missing?

tomaskovacik, 2018-05-28:
please comment out 85:
change 
#define DISC_TRACK_NUMBER_FROM_MPD
to //#define DISC_TRACK_NUMBER_FROM_MPD

compile, upload and check what is on serial port please.

JL01152, 2018-05-28:
commented out line 85.

serial monitor:

Audi Concert I Multimedia Gateway Ver.1.0b
RING

JL01152, 2018-05-28:
solved by changing line 2156.

changed from SetStateIdle()
to SetStatePlay()
and it is working to play music from aux.

tomaskovacik, 2018-05-29:
thx! it should be probably used only with MPD control 

t.

closed 2018-05-29 by tomaskovacik

JL01152, 2018-05-30:
ok soi fried my arduino uno. I somehow shorted some pins while troubleshooting. I have uploaded the code to two different arduino pro minis and the unit does not change to cdc mode now when selecting CD.

Is there a way for me to confirm that I am receiving data from the headunit? Or are arduino pro minis incompatible?

Please help again.

JL01152, 2018-05-30:
i have 16mhz but i will check serial console. so far without unit powered upit keeps repeating the aforementioned messages without the RING.It did not do this with the uno

JL01152, 2018-06-07:
I found my issue. It turned out to be a wrong connection to the ground at the connector. The arduino would seem to power up and work properly but without the the ground it was never going to communicate with the unit. Thanks a million for your support. You are awesome.