tomaskovacik/vwcdavr #7: MPD Controls

JL01152, 2018-06-12:
Ok so I have been messing with the headphone jack on my galaxy s7 edge and I found that I can control the NEXT PREV PLAY/PAUSE functions with just shorting the the MIC ring to the GND ring.

Simulating Button Clicks (shorting to GND)
1 click = play/pause
2 quick clicks = NEXT track
3 quick clicks = PREV track
A 680ohm resistor to gnd simulates pressing the volume down key. ( Media if playing music, Ringtone volume if not playing music)
 
I have yet to find what other resistance values will do. I will try to incorporate these functions using a HEF4066 switch. Hopefully,

My question is. How can I take the command from the Head Unit buttons and have the arduino interpret them into IO functions?

Is this already included in your code?

tomaskovacik, 2018-06-12:
Heh, i dropped headphone button support in avr version when it was at arduino repository  , check history of project files here

https://github.com/tomaskovacik/arduino/commit/2bd01f2e5fd79624dc48ed7dc2a506906b061479

But really go for bluetooth, bk800l, bk3254 or microchip  bm20 module .....

tomaskovacik, 2018-06-13:
Schematics of headset with buttons :

https://forums.androidcentral.com/samsung-galaxy-s4/334540-galaxy-s4-headset-mic-button-wiring-schematic-pinout.html

closed 2018-06-29 by tomaskovacik