tomaskovacik/vwcdavr #12: No CDC detected using ATtiny85

lachaka, 2020-09-28:
Hi,
I can not simulate a CDC using the "cdc_arduino_ported_k9spud_vwcdpic_just_8bit_timers" version. On the serial monitor I got only question marks and the radio says " No CDC". The other version works fine, tested with atmega328.
Tried to slap Clk and Data in, but still no result.

tomaskovacik, 2020-09-28:
attiny85 at 8MHz ?

lachaka, 2020-09-28:
yep

tomaskovacik, 2020-09-28:
try this sketch on atmega328

lachaka, 2020-09-28:
Nothing happens, I even don't get "No CDC"

tomaskovacik, 2020-09-28:
try uncoment //_delay_loop_1(40); on lines 1134 and 1158

lachaka, 2020-09-28:
Now on atmega328 is working, on attiny is the same

tomaskovacik, 2020-09-28:
are you using HW pin 7,6 and 5 ? 
   DataOut   -> 7 (INT0)
   DataIn    -> 6 (PB0)
   Clock     -> 5 (PB1)

lachaka, 2020-09-28:
Yes, but pin 5 is PB0 and 6 is PB1. Anyway after slapping still not success.

tomaskovacik, 2020-09-28:
yep, pb0 is 5, but it strange, anyway I cannot help you more right now, I will test it tomorow
t.

lachaka, 2020-09-28:
No worries. When the radio is powered off I got "00" dataerr and on powering it up another ones but a little bit different, if helpful.

tomaskovacik, 2020-09-29:
I just test code as it was  (no delay when clocking out data) and it works without problem, but I put defined 5microseconds delay there so it is same as 3timer version. Anyway,here is output:
[image]

tomaskovacik, 2020-09-29:
[image]

lachaka, 2020-09-29:
Now the CDC is detected and I got all six CDs, but after clicking two or three times any button the radio blocks and got dataerr in serial monitor. Maybe some other delays are needed.

tomaskovacik, 2020-09-29:
and what you did exactly? , as you can see I got to track 15 so -> 30 commands, got through.

you can play with CLK_DELAY value

lot of times, bad ground connection is problem. check that also

closed 2020-09-30 by tomaskovacik