tomaskovacik/vwcdavr #13: CDC Emulator - sw and hw solution

maro159, 2020-10-05:
Hello,

at the beginning I would like to appreciate your work with VW CDC projects.

I am going to build complete CDC Emulator for VW Gamma IV with aux + bluetooth input for myself. I have been digging in your repositories for a bit.
My plan is to use one of these HW projects [click](https://github.com/tomaskovacik/hw/tree/master/kicad/arduino_based_cdc_emulator)

Would you mind answering a few questions?

-  Is there a significant difference between BM20 and s3860m-s modules (I mean like audio quality and general reliability)?
-  which software should I use to program uC in mentioned projects?

Thanks in advance!

tomaskovacik, 2020-10-05:
do not use s3860m-s, I try it and have to replace it with BK8000L

BM20 is superb, but I only use it with stm32 (the project is in beta stage, but it will be opensource)

I add another one, which uses BK8000L, and it wok, I have few PCBs made for it, the only problem is that I did not swap serial lines, so some jumper wires are required: https://github.com/tomaskovacik/hw/tree/master/kicad/arduino_based_cdc_emulator/atmegax8_with_F-6188

maro159, 2020-10-05:
Thanks for your answer.

- What should _data_out_ pin be connected with? With radio _data_out_ or radio _data_in_? 
- I see, that project for BK8000L does not have 5V voltage regulator? Is it better to have it on different PCB or it casued by PCB minimalization?
-  I also couldn't find _CDC_ENA_12V_ to _CDC_ENA_5V_ converter - isn't it neccesary anymore?
- Is this board ready to be programmed with one of [these](https://github.com/tomaskovacik/vwcdavr/tree/master/CDC_emulator)? Or should I edit it using BK8000L A/T command list?
- What can you say about BK8000L sound quality? Is it sufficient for daily listening or should I better wait for BM20 version?
- One of the projects you have recently added is using CS8645 instead of BK8000L - I guess this one is not tested yet, right?

Generally I am looking for a working CDC emulator with bt module, which supports microphone and controlling _play, pasue, skip, prev_ from the radio panel. I may pay for a one that works - assembled or kit, doesn't really matter.

I am really sorry for asking so many questions. 

Thank you once again!

tomaskovacik, 2020-10-06:
hello,

no problem, it is not refined project, it was more like prototype, but it workout really good :) 

- I added connection notes and fix name of 3pin connector, it is for 7805 switched power supply equivalent (AM1117 goes really hot and also it has huge quiescent current

- I have used cdc_ena_12V for "sw" switching on and off of emulator, but it is really not well documented, so using it as powersource and simply restart mcu is simpliest way, but I used it properly in my vwcdstm32 boards, so I will probably change this design one day and then there will be level shifeter from 12 to 5V

- bk8000L is cheaper then BM20 so this is a cheaper version of emulator with BT, I used it as immediete replacement of OVC board in my 1st prototype and using in in car for 2years, it hase litle bit shitty mic input, but again it is more problem of mic is my guess (I use 1$ mic from aliexpress :D )

if you have time, I can drop you some boards for module into the mail and you can test it :o) unfortunetly, I put all boards for ovc chips into recycle bin so no proper way to test it again. After few years of development of other boards, I am preaty sure problem was the design of the board

regarding SW, I have special version, but it has simply changed pins for dataout,datain and clk, and has changed  output strings on serial line to AT commands, I use this SW as base: 

https://github.com/tomaskovacik/vwcdavr/tree/master/CDC_emulator/cdc_arduino_ported_k9spud_vwcdpic_just_8bit_timers

#define RADIO_COMMAND      PB0 //ICP
#define RADIO_COMMAND_DDR  DDRB
#define RADIO_COMMAND_PORT  PORTB
#define RADIO_COMMAND_PIN PINB
#define RADIO_CLOCK        PD4//PB5
#define RADIO_CLOCK_DDR    DDRD//DDRB
#define RADIO_CLOCK_PORT  PORTD//PORTB
#define RADIO_DATA         PC3//PB3
#define RADIO_DATA_DDR     DDRC//DDRB
#define RADIO_DATA_PORT  PORTC//PORTB

const uint8_t sDATAERR[] PROGMEM = "";
const uint8_t sOVERFLOW[] PROGMEM = "";
const uint8_t sMDISABLE[] PROGMEM = "AT+MC\r\n";
const uint8_t sMENABLE[] PROGMEM = "AT+MA\r\n";
const uint8_t sMINQUIRY[] PROGMEM = "";
const uint8_t sPRV_LIST[] PROGMEM = "AT+MA\r\n";
const uint8_t sNXT_LIST[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST1[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST2[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST3[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST4[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST5[] PROGMEM = "AT+MA\r\n";
const uint8_t sLIST6[] PROGMEM = "AT+MA\r\n";
const uint8_t sRANDOM[] PROGMEM = "";
const uint8_t sPLAY[] PROGMEM = "AT+MA\r\n";
const uint8_t sSCAN[] PROGMEM = "";
const uint8_t sSTOP[] PROGMEM = "AT+MC\r\n";
const uint8_t sNEXT[] PROGMEM = "AT+MD\r\n";
const uint8_t sPREVIOUS[] PROGMEM = "AT+ME\r\n";
const uint8_t sRING[] PROGMEM = "";
const uint8_t sIDENTIFY[] PROGMEM = "";
const uint8_t sNEWLINE[] PROGMEM = "";
const uint8_t sDASH[] PROGMEM = "";
const uint8_t sHEX[] PROGMEM = "";
const uint8_t sVERSION[] PROGMEM = "";

tomaskovacik, 2020-10-06:
HW part moved to this repository https://github.com/tomaskovacik/vwcdavr/tree/master/HW/arduino_based_cdc_emulator

maro159, 2020-10-07:
Thank you.
Can I use _CDC_ENA_ (+12V cd changer supply switched by radio) to power the whole circuit? I would connect both _+12V_ and _CDC_ENA_12V_ pins to radio switched +12V available on radio C connector. I think current of the circuit with atmega, bt module and amplifier won't exceed max current of that radio output. What do you think?

I would appreciate getting some PCBs from you and yes, I have some time to test it and write about my results. How can I DM you?

tomaskovacik, 2020-10-07:
yes you can drive the whole circuit from switched 12V, but I cannot say if all radios have the same current capability, old Audi concert/chorus1 have a chip which max at 3A
you can drop me email at nail@nodomain.sk

closed 2020-10-17 by tomaskovacik