tomaskovacik/vwcdavr #15: I think I need some help

DaMusikant, 2021-02-02:
Hi there,

at first a really big commendation for your work.
Since a few hours I am trying to understand the cdc_arduino_ported_k9spud_vwcdpic programm. 
I use the arduino nano right now. By simple copy and paste the program into the Arduino IDE and check it the compiler says:

`C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic\cdc_arduino_ported_k9spud_vwcdpic.ino:361:13: warning: 'void ScanCommandBytes()' used but never defined

 static void ScanCommandBytes(void);

             ^~~~~~~~~~~~~~~~

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic\cdc_arduino_ported_k9spud_vwcdpic.ino:382:13: warning: 'void SendPacket()' used but never defined

 static void SendPacket(void);

             ^~~~~~~~~~

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic\cdc_arduino_ported_k9spud_vwcdpic.ino:375:13: warning: 'void EnqueueString(const uint8_t*)' used but never defined

 static void EnqueueString(const uint8_t *addr PROGMEM);

             ^~~~~~~~~~~~~

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic\cdc_arduino_ported_k9spud_vwcdpic.ino:365:13: warning: 'void SetStateIdle()' used but never defined

 static void SetStateIdle(void);

             ^~~~~~~~~~~~

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic\cdc_arduino_ported_k9spud_vwcdpic.ino:377:13: warning: 'void ResetTime()' used but never defined

 static void ResetTime(void);

             ^~~~~~~~~

Compiling libraries...

Compiling core...

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:455: undefined reference to `ResetTime()'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:456: undefined reference to `SetStateIdle()'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:458: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:459: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:460: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:461: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:480: undefined reference to `SendPacket()'

C:\Users\Johannes\AppData\Local\Temp\cchlkGFH.ltrans0.ltrans.o: In function `CDC_Protocol':

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:832: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:838: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:842: undefined reference to `ScanCommandBytes()'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:799: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:800: undefined reference to `EnqueueString(unsigned char const*)'

C:\Users\Johannes\Documents\Arduino\vag_cdcFaker\cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino:801: undefined reference to `EnqueueString(unsigned char const*)'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error at compyling.`

Those are the most important things I think. Because i'm relativ new to Ardunio I'm out whats wrong there. In C# I would think that the variable is not definied, but that should already happend, or? And the second part should be a consequential error.

What do you think?

Thank you very much and greetings,
DaMusikant

tomaskovacik, 2021-02-02:
put functions.ino into same folder as cdc_arduino_ported_k9spud_vwcdpic.ino

DaMusikant, 2021-02-02:
Wow, that was easy. 
Your great, thank you :)

tomaskovacik, 2021-02-02:
no problem :)

closed 2021-02-02 by tomaskovacik