tomaskovacik/vwcdavr #16: Serial Print and theyre meanings

DaMusikant, 2021-02-16:
Hi,
you already helped me lot but I'm again stuck in the middle of the code.

The problem is, that the Arduino is sending some messages that are not encodible. Every third message I got a "007904FB" if I am in CD Changer mode with my radio and "007901FE" if I am not in CD Changer mode. 

If I press down the button to go into CD Changer mode I got 

`MENABLE007904FBMENABLE007904FBMENABLEMENABLE007904FBMENABLE`

I think those are current status updates. But the problem is that they are also going in between the main messages like `007904FBLIST2` 

For my purpose I don't need those messages but I didn't found any possibility to disable them.

Can you help me? 

Thank you and greetings

tomaskovacik, 2021-02-16:
commands are decoded here: https://github.com/tomaskovacik/vwcdavr/blob/13e3b846cec330603b42a533d24978976087eb6e/CDC_emulator/cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino#L895

but only in 1st byte is 0x53 2nd is 0x2c :+1: 
https://github.com/tomaskovacik/vwcdavr/blob/13e3b846cec330603b42a533d24978976087eb6e/CDC_emulator/cdc_arduino_ported_k9spud_vwcdpic/functions.ino#L14

put some debug info there and check if it is dumped in ScanCommandBytes() with DumpFullCommand() or it is dumped in DecodeCommand()

but I think it is wrong command send by radio and ScanCommandBytes() did not understand it, because DecodeCommand() should mark dash before unknown cmd and newline after this cmd.

DaMusikant, 2021-08-05:
Hi, after a long period of try and error I still stuck. I'm new to coding and all that stuff, thats properly the main problem ;)
I tried to insert some debug infos but nothing changed, or the whole code was getting faulty. 
To my understand, the ScanCommandBytes checks if the commands are correct. If not it is waiting until a good one is arriving. Right?

Another idea (?): It is working almost fine, the only problem are those commands. So if it is possible to just throw them into the bin and use only the ones, that we are understand, it would work fine. But on the search for this possibility I got also stuck.

Can you help another time? Thank you

tomaskovacik, 2021-08-05:
standard code should output each 10s  on the serial terminal "Audi Concert I Multimedia Gateway Ver." 

did you get this text?

DaMusikant, 2021-08-05:
I get the text only every startup of the whole system.
The message pops up every 4th second btw, no matter in which mode (CDC or FM) I am

tomaskovacik, 2021-08-05:
That is ok i have to look into it

DaMusikant, 2021-08-05:
Thank you very much :)

DaMusikant, 2021-08-05:
Hi, I found another method to avoid the problem, just ignore the commands on the other end of the serial end. But thank you very much :)