tomaskovacik/vwcdavr #18: Question: Receiving media keys/ multi stearing commands and pass to CDC emulator

istals, 2022-04-07:
Hey!

I have a question, 

Is there a possibility to get prev/next commands from the head unit and multi steering next/prev track signals pass to CDC emulators?

tomaskovacik, 2022-04-07:
these?

https://github.com/tomaskovacik/vwcdavr/blob/eaf4728656153876310d7062e409dd500cb984b5/CDC_emulator/cdc_arduino_ported_k9spud_vwcdpic/cdc_arduino_ported_k9spud_vwcdpic.ino#L209

tomaskovacik, 2022-04-07:
what exactly do you need to do with them?  if you press button on MFSW or on radio then emulator will change track

tomaskovacik, 2022-04-07:
is it working without your changes ?

istals, 2022-04-07:
Regarding this, how could I test that these buttons are working and I'm receiving those bytes?
DUMPMODE and DUMPMODE2 must be enabled?

istals, 2022-04-07:
No, not working no data from head unit

tomaskovacik, 2022-04-07:
arduino pin 8 is connected to dataout on the radio?

   RADIO PIN -> arduino pin
   DataOut   -> digital 8 (ICP1)
   DataIn    -> digital 11(PB3) (10 for atmega328PB)
   Clock     -> digital 13(PB5)

tomaskovacik, 2022-04-07:
you should see some commands on serial at startup of radio, mainly disable and enable, depends on state in which radio started

istals, 2022-04-07:
Looks like my head unit is not providing any data, closing

closed 2022-04-07 by istals

tomaskovacik, 2022-04-07:
radio must be coded to have cdc enabled, try it 

http://wiki.ross-tech.com/wiki/index.php/Audi_Radio_Generation_I#Coding