tomaskovacik/vwcdavr #21: ATtiny 2313 Support

tengzl33t, 2024-08-01:
Hello, can it be ported to ATtiny 2313 with Blaupunkt MCD RNS button support added?
There is already existing firmware for it, but without button support and no source code available.

RNS button actions -> ATtiny -> Bluetooth module button inputs

Thank you.
[Emulator MCD.zip](https://github.com/user-attachments/files/16454051/Emulator.MCD.zip)

tomaskovacik, 2024-08-05:
[attiny2313_4313.zip](https://github.com/user-attachments/files/16491360/attiny2313_4313.zip)
I did a quick look at datasheet and add som defines to select proper registers:
```
|| defined(__AVR_ATtiny4313__)
```
try attached code, pin outputs are defined here:

```
#if defined(__AVR_ATtiny85__) || defined(__AVR_ATtiny45__) || defined(__AVR_ATtiny4313__)
#define RADIO_COMMAND      PB2
#define RADIO_COMMAND_DDR  DDRB
#define RADIO_COMMAND_PORT PORTB
#define RADIO_COMMAND_PIN  PINB
#define RADIO_CLOCK        PB1
#define RADIO_CLOCK_DDR    DDRB
#define RADIO_CLOCK_PORT   PORTB
#define RADIO_CLOCK_PIN    PINB
#define RADIO_DATA         PB0
#define RADIO_DATA_DDR     DDRB
#define RADIO_DATA_PORT    PORTB
```

tengzl33t, 2024-08-12:
Unfortunately, I was not able to get any replies from some of the potential devs of that firmware. So looks like it is not possible to get any sources now.

Is it possible to maintain the same pins in use, like how they are used in the project I sent in the first message to not rebuild the board?

I have also mentioned forwarding the radio button actions to the Bluetooth module. I think this could be done with converting the codes got from radio to 3.3V High levels on desired pins.

Thank you.

tomaskovacik, 2024-08-13:
well, I changes pins on attached firmware, you should really do it your self, it was easy as changing 16 pins definitions which I wrote you in prev. comment ... 
regarding controlling bluetooth module, that is out of my reach, I do not know which module you have, I do not have that radio ....  so it is up to you :) 
[firm.zip](https://github.com/user-attachments/files/16596988/firm.zip)

closed 2024-11-02 by tengzl33t